vintagepc/MINI404 #146: [BUG]  MK3.5 bootloader does not work.

vintagepc, 2024-03-03:
Per title, trying to run the MK3.5 v5.2.1 firmware using the .bbf file fails to run; the bootloader is not functional in the sim .

BlueFyre, 2024-03-03:
Per #145:
> Unfortunately that means even if you do get past the GPIO error, you'd need to have a locally built "noboot" ELF or BIN file built from the Prusa repository rather than a BBF at this time. Since the bootloader is closed source, I don't know when or if I'll be able to fix that. Sorry.

I imagine that's different than just the [regular bootloader](https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/master/utils/bootstrap.py#L81) from Prusa? Does that also apply to the MK4?

vintagepc, 2024-03-03:
The file you linked is the one I retrieved for my test with the official 5.2.1 BBF and resulted in this issue. 

As far as I know the MK4 bootloader and .bbf combo works fine; the 3.5 is a relative newcomer here and so I'm not sure what is different yet. Based on the early firmware examinations when I added the 3.5 support (most of my work here is based on reverse-engineering the firmware sources if schematics are not available) it's not all that different from the Mk4 board - it's actually defined in the mk4 machine [here](https://github.com/vintagepc/MINI404/blob/fe0eb2d6069cf20705ccd065d42f3e116eabb066/hw/arm/prusa/prusa-mk4.c#L238) just like the 3.9.

That said, there are no entries or pictures of this board up in the e-shop so I can't say for certain how "different" it is or what would cause this, and the e-shop doesn't list the xbuddy board as compatible with the 3.5, so *something* must be different. 

The "noboot" firmware is a binary that doesn't require the bootloader (nor expect it) and is located in a different region in the printer flash space so it can be booted directly. I did confirm I was able to boot and start that (though it does complain the hotend fan isn't spinning, so there may be more changes I need to make.)

vintagepc, 2024-03-03:
[image]

Figured it out, there was a stupid ordering bug with how the GPIO pins were being initialized, that led to the display data being bad and the boot not able to proceed.

BlueFyre, 2024-03-03:
Awesome! I do have both MK3.5 and MK4. It's the same board as far as I can tell. From when I was looking through the code the only difference is the detection based on the presence of the loveboard (among other checks) and then they do now have different firmware files but you can be shipped the wrong firmware on it (which then would complain and tell you to flash the right one)

vintagepc, 2024-03-03:
Yep, if there is no loveboard there's some additional detection that happens and this was bugged. 

Just need to fix the unit tests I broke now  before I can merge that :grin:

PR #149 (2024-03-03): Misc Mk3.5 fixes
- Fix MK3.5 boot with BBF

- Fixes #146
commit e9f167df (2024-03-03): Misc Mk3.5 fixes (#149)

* Fix bug in GPIO initialization sequence. Fixes #146

* Adjust failing unit tests
closed 2024-03-03 by vintagepc